wavelab/libwave #22: Compile Error using supplied script

arunxdas, 2017-03-23:
tried to install the deps using the supplied script on my desktop. I got the following error when trying to install opencv

nvcc fatal   : Unsupported gpu architecture 'compute_11'
CMake Error at cuda_compile_generated_gpu_mat.cu.o.cmake:208 (message):
  Error generating
  /usr/local/src/opencv/opencv-3.0.0-alpha/build/modules/core/CMakeFiles/cuda_compile.dir/src/cuda/./cuda_compile_generated_gpu_mat.cu.o

Are we enabling cuda by default?

arunxdas, 2017-03-23:
Here is the full error:

`[ 36%] Building NVCC (Device) object modules/core/CMakeFiles/cuda_compile.dir/src/cuda/cuda_compile_generated_gpu_mat.cu.o
CMake Warning (dev) at cuda_compile_generated_gpu_mat.cu.o.cmake:139 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted keywords like "COMMAND" will no longer be interpreted as keywords
  when the policy is set to NEW.  Since the policy is not set the OLD
  behavior will be used.
Call Stack (most recent call first):
  cuda_compile_generated_gpu_mat.cu.o.cmake:166 (cuda_execute_process)
  
This warning is for project developers.  Use -Wno-dev to suppress it.

nvcc fatal   : Unsupported gpu architecture 'compute_11'
CMake Error at cuda_compile_generated_gpu_mat.cu.o.cmake:208 (message):
  Error generating
  /usr/local/src/opencv/opencv-3.0.0-alpha/build/modules/core/CMakeFiles/cuda_compile.dir/src/cuda/./cuda_compile_generated_gpu_mat.cu.o

modules/core/CMakeFiles/opencv_core.dir/build.make:63: recipe for target 'modules/core/CMakeFiles/cuda_compile.dir/src/cuda/cuda_compile_generated_gpu_mat.cu.o' failed
make[2]: *** [modules/core/CMakeFiles/cuda_compile.dir/src/cuda/cuda_compile_generated_gpu_mat.cu.o] Error 1
CMakeFiles/Makefile2:1300: recipe for target 'modules/core/CMakeFiles/opencv_core.dir/all' failed
make[1]: *** [modules/core/CMakeFiles/opencv_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 36%] Built target pch_Generate_opencv_videostab
[ 36%] Built target pch_Generate_opencv_test_viz
[ 36%] Built target pch_Generate_opencv_viz
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
`

chutsu, 2017-03-23:
Nope, we are not enabling cuda by default. Further, I dropped opencv 3.0 in PR #21

chutsu, 2017-03-23:
also by installing opencv 3.0 I might have broken your computer :D, you coming in? I can fix it for you

arunxdas, 2017-03-23:
I am working on my desktop at home.  the compile for opencv failed half way, but I was still able to build libwave.  Should the default libwave install script be breaking people's computers? :smile:

chutsu, 2017-03-23:
No, thats why I'm trying to get this PR #21 in :D

closed 2017-03-23 by chutsu