wavelab/libwave #45: Properly find dependencies with CMake

leokoppel, 2017-04-13:
The purpose of CMake is to allow cross-platform builds without making assumptions about directory structure.

Over time some absolute paths have crept into CMakeLists files. Conversely, some dependencies are found by virtue of being in the default path, but are never declared. These practices may cause problems for users with different directory structures, different OS, multiple versions of libraries installed, etc.

The tasks for this issue are:

#### Remove hardcoded paths from CMake files
   This includes lines such as
   ```cmake
   INCLUDE_DIRECTORIES(/usr/include/eigen3)
   ```
   and 
   ```cmake
   INCLUDE_DIRECTORIES(/usr/include)
   ```

#### Explicitly list all dependencies
   For example, Boost is not listed

leokoppel, 2017-04-13:
In case its useful, here are my slides on where `find_package` looks for things: https://docs.google.com/presentation/d/1zyuT8qPTHDNuhl5jSF2-NTLZ8PPnuNF0VHDxaFFHuPM/edit?usp=sharing

jitrc, 2017-04-17:
Yes and I use this for Eigen
```
find_package(Eigen3 QUIET)
if(NOT EIGEN3_FOUND)
  # Fallback to cmake_modules
  find_package(cmake_modules REQUIRED)
  find_package(Eigen REQUIRED)
  set(EIGEN3_INCLUDE_DIRS ${EIGEN_INCLUDE_DIRS})
  set(EIGEN3_LIBRARIES ${EIGEN_LIBRARIES})  # Not strictly necessary as Eigen is head only
  # Possibly map additional variables to the EIGEN3_ prefix.
else()
  set(EIGEN3_INCLUDE_DIRS ${EIGEN3_INCLUDE_DIR})
endif()
```

chutsu, 2017-04-17:
Sure go ahead, there are already 2 (Kindr and Eigen cmake `Find` files in the `libwave/cmake` dir), but fully agree on listing all dependencies

chutsu, 2017-05-24:
Consensus has been made, we should use cmake files to find dependencies, closing this now but should create new issues for independent issues for each new cmake dependency config file, e.g. PCL, Ceres.

leokoppel, 2017-05-24:
This issue is a bit broad but it's pretty common to have an overarching issue which is mentioned in sub-issues or multiple PRs. I see it as still open.

Note the issue description says nothing about the creation of "cmake dependency config files". That is not inherent to the issue.

> leokoppel added a commit that referenced this issue 9 hours ago

Oh, that repetitive github referencing is ugly. I did not know it prints the message every time I rebase a commit!

chutsu, 2017-05-24:
ok, keeping it open.

PR #225 (2017-10-22): Refactor CMake dependencies
Resolves #45

Link against _imported targets_ and set _properties_ instead of using INCLUDE_DIRECTORIES and the like. This method is more explicit, and makes dependencies part of the _link interface_ of each module. 

See https://rix0r.nl/blog/2015/08/13/cmake-guide/

For example: if I link a target to wave_utils, yaml-cpp is automatically linked. Also, wave_utils/include is automatically added to  `INCLUDE_DIRECTORIES` for that target.

There is no more need to `INCLUDE_DIRECTORIES` everything at the top level, or to explicitly call `ADD_DEPENDENCIES` at the top le [... cut]
commit 7da8bdc1 (2017-10-22): Refactor CMake dependencies (#225)

Resolves #45

Link against imported targets and set properties instead of using INCLUDE_DIRECTORIES and the like. This method is more explicit, and makes dependencies part of the link interface of each module.

See https://rix0r.nl/blog/2015/08/13/cmake-guide/

* Explicitly set link dependencies and interface each module

Use modern, explicit commands such as TARGET_INCLUDE_DIRECTORIES instead of
INCLUDE_DIRECTORIES.

Do not define inter-module dependencies in top-level CMakeLists. Instead, each
module defines its own.

Reference: https://rix0r.nl/blog/2015/ [... cut]
closed 2017-10-22 by leokoppel