wavelab/libwave #65: Create front end for vision-based SLAM

navganti, 2017-05-12:
The goal of this issue is to develop the front end for vision based (VO/VIO) SLAM using libwave.

chutsu, 2017-05-14:
I've done that a while ago: https://github.com/chutsu/yarl/blob/master/src/vision/fast.cpp

What you may want to focus instead is a generic front end interface, for example given an image, how would you write the front end "classes" to easily switch between different features (be it Good, Fast, FREAK, etc...) and how would the feature correspondence look like?

navganti, 2017-05-15:
@chutsu sounds good. I started the exception stuff; I'll start migrating your yarl code into libwave then and rename the issue.

navganti, 2017-05-15:
@chutsu, yarl is released under GPLv3. I'm not sure what kind of license we want for libwave, but this could lead to some problems. To avoid these problems maybe I should write our own, and not rely on other code...?

tagging @michaelhsmart @leokoppel @arunxdas to also chime in on this.

chutsu, 2017-05-15:
yarl is mine, you can do whatever you want.

navganti, 2017-05-15:
Haha I figured you wouldn't sue us. @michaelhsmart mentioned it was more of a concern with regards to our industry partners, and the work we do with them.

leokoppel, 2017-05-15:
Good point. The idea is to copy parts of the code into this library, right, not include it as a dependency? Then it would fall under this license. Chris is allowed to re-license his own code.

But it does get complicated. For example, if a partner runs some kind of IP analysis on the code, it may flag Chris' other library as a plagiarized source. I don't know if that's a problem.

michaelhsmart, 2017-05-15:
Do we have a private location for this discussion? Our internal licensing issues should not be on public issues :)

navganti, 2017-05-15:
To Basecamp we go!

commit 12a35e0a (2017-05-25): Formatting fixes

Resolves #65

Runs clang-format after it was not run for #84

Related #93, #101
PR #103 (2017-05-25): Formatting fixes
Resolves #65

Runs clang-format after it was not run for #84

Related #93, #101
commit d4c1ea2f (2017-05-25): Formatting fixes (#103)

Resolves #65

Runs clang-format after it was not run for #84

Related #93, #101
closed 2017-05-25 by navganti