wavelab/libwave #68: Modify install_deps to install OpenCV3

navganti, 2017-05-13:
Currently installation script installs OpenCV2.4. 

Decision as per discussion here: https://public.3.basecamp.com/p/3uFf3ZrDSChhjESrMj8Lh4vq

chutsu, 2017-05-14:
I know we discussed it in basecamp, but I'm not sure code related things should be discussed in basecamp and linked from issue to basecamp. Image this, if we used something else other than basecamp the discussion details will be losted.

navganti, 2017-05-15:
Good point @chutsu. Figured since Basecamp is our main "discussion" hub it's where we should chat about it. I didn't want to create a Github Issue without first discussing the concern with the group, especially since this could affect quite a bit of people, and isn't just a code specific. 

I guess in the future I'll create the Github Issue, any discussion could be had here, and THEN the issue link can be posted in Basecamp/Slack so everyone on the project sees it with some visibility.

chutsu, 2017-05-15:
Fix in #69

closed 2017-05-15 by chutsu