wavelab/libwave #76: Travis cache failing

leokoppel, 2017-05-18:
Travis is printing:
```
found cache
adding /usr/local/src to cache
running `casher add/usr/local/src` took longer than 180 seconds and has been aborted
```

As a result, ceres and opencv build are still really slow.

Unknown whether the cache is actually so big that it takes >180s, or whether it simply hangs for some other reason.

Related: #14

chutsu, 2017-05-24:
culprit is during caching, it takes too long for travis to add `/usr/local/src` to cache

[image]

leokoppel, 2017-05-24:
Yes, that's the output I pasted in the issue description : )

leokoppel, 2017-06-03:
Travis cache is not relevant now that we have a better solution, PPAs. See #106 #132

closed 2017-06-03 by leokoppel