wavelab/libwave #90: Reconfigure Travis notifcations to slack instead of emailing Chris

chutsu, 2017-05-23:
At current whenever a branch fails or gets fixed @chutsu gets notified. This issue aims to

- Remove Chris from travis notications
- Redirect travis notifications to slack

leokoppel, 2017-05-24:
It would be nice to limit this so it doesn't show build failures on topic branches on slack. Ideally it would notify only for "cron" builds, so we have an alert if master is not building due to some changed depedency.

Unfortunately there's no way to limit it to a specific branch through travis.yml: https://github.com/travis-ci/travis-ci/issues/1405

Maybe:
 - disable notifications
 - use a separate slack channel?

chutsu, 2017-05-24:
Actually I'm for disabling notifications actually

commit dfc468b6 (2017-05-24): Merge pull request #95 from wavelab/fix_90_notifications

Resolves #90 

As Leo already mentioned in PR #91, the travis docs says:

By default, email notifications are sent to the committer and the commit author when they are members of the repository
This PR adds:

notifications:
    emails: false
to completely disable email notifications.
closed 2017-05-24 by leokoppel