wavelab/libwave #100: Create cpplint configuration file for libwave

chutsu, 2017-05-25:


leokoppel, 2017-05-25:
How useful are the cpplint checks for us (the ones not already handled by clang-format)?

If I run cpplint now the warnings are
 
 29 [legal/copyright]
 25 [readability/namespace]
 19 [build/include_what_you_use]  (IMO excessive - my IDE often puts a contradictory warning that the includes are redundant)
 10 [runtime/references] (IMO not useful)
  7 [whitespace/line_length] (lines surrounded with "clang-format off")
  5 [whitespace/parens]
  3 [readability/casting]
  2 [build/include_order]

navganti, 2017-05-25:
I agree with @leokoppel about some being excessive. Primarily the include_what_you_use...that one is always overkill

chutsu, 2017-05-25:
I think I may have a better question, is there a Static Code Analyzer we should be using that helps prevent hidden C++ potential pitfalls that the compiler warnings don't tell us?

yuzhangbit, 2017-05-25:
cppcheck?

leokoppel, 2017-05-25:
[clang-tidy](http://clang.llvm.org/extra/clang-tidy/) - includes linter and analyzer checks

cglwn, 2017-05-26:
I've used [Coverity](https://scan.coverity.com/github) and it's pretty good with uncovering actual bugs.

chutsu, 2017-05-26:
@cglwn, if I can I'll probably opt for a solution that doesn't require us to setup a webservice to run code static analyzer.

cglwn, 2017-05-26:
That's fair. Can't say whether it was used at work, but the Waterloo software quality course I took (SE 465) had a project with it where we analyzed an open source library with it and it worked really well.

I can also vouch for clang-tidy being good and probably the best bang-for-buck.

The standalone [include-what-you-use](https://include-what-you-use.org/) is also a lot better than the one with cpplint and can be really helpful for keeping down compile times and debugging missing symbols because of changes in upstream headers.

closed 2018-01-16 by chutsu