wavelab/libwave #106: Demonstrate how to create a Ubuntu deb package with OpenCV3

chutsu, 2017-05-25:
At current our travis takes 20 minutes+ to install opencv, we need a faster way to install libwave dependencies. 1 solution is to install big dependencies like OpenCV3 as a Ubuntu debian package, this issue demonstrates how one can create such a package.

archielee, 2017-05-26:
At the moment, why does it take 15 minutes to complete the installation?

archielee, 2017-05-26:
So I found this link:
https://unix.stackexchange.com/questions/46122/how-can-i-create-a-deb-package-with-my-compiled-opencv-build

It looks like you can already create debs without any special packaging. It requires a small change (cloning from source and checking out tags/3.2.0). The CMake flags can be manually set in CMakeCache.txt

archielee, 2017-05-26:
note: it could just be as simple as adding a line in the install script:

```
    cmake \
        -D CMAKE_BUILD_TYPE=RELEASE \
        -D CMAKE_INSTALL_PREFIX=/usr/local \
        -D WITH_TBB=ON \
        -D WITH_V4L=ON \
        -D WITH_QT=ON \
        -D WITH_OPENGL=ON \
        -D CPACK_BINARY_DEB=ON ..
```

and running `make package -j$(nproc)`

It should generate 3 debs: lib, dev, python

archielee, 2017-05-26:
If we want to use a PPA, we cannot upload .deb packages directly. They must be built on Launchpad. A guide is here:
https://askubuntu.com/questions/71199/manually-created-deb-how-do-i-upload-to-a-ppa

archielee, 2017-05-26:
I uploaded a test package onto my personal PPA, it's currently processing but you can test with that. I'm also going to push what I have onto the branch. I'll write up a small README to explain how to build it

archielee, 2017-05-26:
Made an initial commit on the branch but it's huge. There's a readme in scripts/install/opencv3/debian

leokoppel, 2017-05-26:
Cool, nice work @archielee 

Just checking, we will not keep it in this repo right? It is not really the place for 160MB of opencv.

chutsu, 2017-05-26:
No, I just reminded him to remove the debs.

archielee, 2017-05-27:
I won't be around next week but will likely have internet access. Let me know if you need me to remove anything else

PR #133 (2017-06-03): Install opencv3 from ppa
Resolves #106 

I created a debian package for opencv3.2.0 with the [configuration here](https://github.com/leokoppel/opencv/tree/3.2.0-trusty/debian), based on the [Debian opencv 3.1.0 experimental package](https://packages.debian.org/source/experimental/opencv). It is currently built on [my ppa](https://launchpad.net/~lkoppel/+archive/ubuntu/opencv) for both trusty and xenial.

This package includes most opencv modules (e.g. no gstreamer or matlab support) and almost all opencv_contrib modules (no dnn or tracking). Maybe should cut down what is installed on travis to [... cut]
closed 2017-06-03 by chutsu