wavelab/libwave #171: Autoformatting Questions/Discussion

michaelhsmart, 2017-07-06:
As most of you may know, I'm looking at how to abstract the autoformatting used here in a way we can use across all of our projects. Consequently, I've been reviewing the `.clang-format` file we're using here.

[Reference to ClangFormatStyleOptions](https://clang.llvm.org/docs/ClangFormatStyleOptions.html)

I've got some questions/comments about it:

1) The base style is currently: `BasedOnStyle: Chromium`. If we change this to `BasedOnStyle: Google` the only change I've observed on other projects is that `clang-format` will re-organize the order of `#include` statements (possibly to satisfy cpplint's frequent complaints?). Are there any objections to switching this? I spoke with @chutsu and this setting is just a legacy of the configuration's early development for his individual projects. Note that if we don't want to atuomatically order includes, we can switch that off with a specific parameter (`SortIncludes:false`).

2) `AllowShortCaseLabelsOnASingleLine: true`.
Does anyone care if we change this to false? I personally prefer

```cpp
switch (a) {
    case 1:
        x = 1;
        break;
    case 2:
        return;
}
```
over
```cpp
switch (a) {
    case 1: x = 1; break;
    case 2: return;
}
```

I like `return` and `break` statements to appear on their own lines as it makes them pop out more to me. Note that this parameter is also the only `AllowShort*` parameter set to true:
```yaml
AllowShortBlocksOnASingleLine: false
AllowShortCaseLabelsOnASingleLine: true
AllowShortFunctionsOnASingleLine: Empty
AllowShortIfStatementsOnASingleLine: false
AllowShortLoopsOnASingleLine: false
```

3) We currently have `ExperimentalAutoDetectBinPacking: true`. The docs for this parameter:

> If true, clang-format detects whether function calls and definitions are formatted with one parameter per line.
> 
> Each call can be bin-packed, one-per-line or inconclusive. If it is inconclusive, e.g. completely on one line, but a decision needs to be made, clang-format analyzes whether there are other bin-packed cases in the input file and act accordingly.
> 
> NOTE: This is an experimental flag, that might go away or be renamed. Do not use this in config files, etc. Use at your own risk.

Are there any objections to removing this setting? We already set the other BinPacking parameters explicitly, so it is unclear that this is even used.

4) Is there a reason we are using `clang-format-3.8`? The newer(est?) version allows setting `PenaltyBreakAssignment` so that we would potentially be able to get rid of these style changes:

```
-    this->some_var = this->public_nh_.subscribe(
-      "blahblah", 1, &SuperFooBarNodelet::blahblahCb, this);
+    this->some_var =
+      this->public_nh_.subscribe("blahblah", 1, &SuperFooBarNodelet::blahblahCb, this);
```

We're never going to be able to get perfect autoformatting, but I don't have all of the available tools to play with in `3.8` for line breaking. I'm not opposed to walking away from it as "good enough" but if there is an available parameter that would make things nicer, it seems a shame/waste to not try it.

5) Are there any objections to moving to a 90 column limit? It's still small enough to support two windows and a sidebar arrayed horizontally.

END: Since this is all stylistic/aesthetic, please remember:
* concrete/technical reasons are important(dominant) to mention if you have them
* preferences are subjective
* play nice :smile:

leokoppel, 2017-07-06:
1. >  re-organize the order of #include statements 

I don't know if this is good. First it is not always correct, for example, I think it treats ros headers (which have a `.h` extension) as "C system headers" even though they are not. Second, it orders headers within each section alphabetically, but we may want to group them in some other way (e.g, group ros messages together).

2. > Does anyone care if we change this to false? I personally prefer

Sure, seems OK

3. >Are there any objections to removing this setting?

No

4. >Is there a reason we are using clang-format-3.8
It's the version on ubuntu xenial (and available on travis).

Apparently `PenaltyBreakAssignment` was only added to their trunk a month ago, so it's not in any stable release. Sorry I did not realize this when I mentioned it.

5. > Are there any objections to moving to a 90 column limit?

Not here : )

michaelhsmart, 2017-07-06:
Regarding (1), I was unclear (apologies): We can
* A) Stick with as is (there is no other mention to Chromium ever anywhere as far as I can find, whereas we do mention a basis on Google in some projects)
* B) Switch to Google (defaults to include sorting)
* C) Switch to Google and explicitly disable sorting.

From your comment, @leokoppel, I assume (C) is ok?

leokoppel, 2017-07-06:
Yep. by the way here is the difference between Google and Chromium style (https://github.com/llvm-mirror/clang/blob/release_38/lib/Format/Format.cpp#L597)

```cpp
ChromiumStyle.AllowAllParametersOfDeclarationOnNextLine = false;
ChromiumStyle.AllowShortFunctionsOnASingleLine = FormatStyle::SFS_Inline;
ChromiumStyle.AllowShortIfStatementsOnASingleLine = false;
ChromiumStyle.AllowShortLoopsOnASingleLine = false;
ChromiumStyle.BinPackParameters = false;
ChromiumStyle.DerivePointerAlignment = false;
ChromiumStyle.SortIncludes = false;
```

michaelhsmart, 2017-07-06:
@leokoppel thanks!

chutsu, 2017-07-11:
Ok, very late to the party. Just to say that I'm in-line with Leo on (1), we should not allow auto sorting of headers, I just had a problem where header file declaration determines whether AprilTags built or not. Luckily it was a 1 liner, but for more complex third party libraries where header file include order matters this can be a nightmare.

chutsu, 2017-07-11:
If we are all in agreement shall we close this @michaelhsmart?

closed 2017-07-11 by michaelhsmart

leokoppel, 2017-07-11:
@chutsu do you want to make matching changes to libwave repo?

chutsu, 2017-07-11:
Sure, lets wait for Mike to finish finalizing the clang-format file and then we can steal it :D