wavelab/libwave #199: Ceres 1.13 compiled with wrong Eigen version - Libwave installation broken

navganti, 2017-07-29:
#### Platform
Bug reproduced consistently on my machine. Lenovo Thinkpad P50, running Ubuntu 16.04 Xenial.

#### Steps to Reproduce
1. Follow libwave installation procedure (either @chutsu's curl script, or the manual installation).

#### Expected result
Libwave builds with no problems.

#### Observed result
CMake error.

CMake Error at /usr/local/lib/cmake/Ceres/CeresConfig.cmake:88 (message):
  Failed to find Ceres - Found Eigen dependency, but the version of Eigen
  found (3.3.2) does not exactly match the version of Eigen Ceres was
  compiled with (3.2.92).  This can cause subtle bugs by triggering
  violations of the One Definition Rule.  See the Wikipedia article
  http://en.wikipedia.org/wiki/One_Definition_Rule for more details

#### Cause

Eigen and Ceres were updated in the ppa, but Ceres was compiled with a previous version of Eigen. This causes an error in `CeresConfig.cmake`, since `Ceres_FOUND` is now `false`. 

#### Related issues

leokoppel, 2017-07-29:
The ppa installs everything to `/usr`. The ceres in `/usr/local` is your own installation. Remove it.

chutsu, 2017-07-29:
@navganti in specific remove the `/usr/local/lib/cmake/Ceres/CeresConfig.cmake`

chutsu, 2017-07-29:
Which reminds me, we should really have an uninstall cmake target :D

navganti, 2017-07-29:
@leokoppel sorry! Forgot that I installed Ceres myself a while ago, and didn't realize the new one installs to `/usr`. Thanks :slightly_smiling_face: 

@chutsu I also had to delete my Ceres folder within `/usr/local/share`. All is good now.

closed 2017-07-29 by navganti

leokoppel, 2017-07-29:
Thanks for documenting, probably others will have this conflict

`/sudo rm -r /usr/local/include/ceres /usr/local/lib/cmake/Ceres /usr/lib/libceres.*`