wavelab/libwave #267: readImageSequence method elicits build complaints in catkin (opencv versioning issue)

navganti, 2018-01-09:


navganti, 2018-01-09:
Okay so @leokoppel. Regarding this, we're having issues in numerous places - as you recommended to me like a month ago, I only got it working once I removed my custom installation of OpenCV. Due to ROS's ties with OpenCV in a bunch of places I figured it was best to not go near that.

navganti, 2018-01-09:
@leokoppel I don't really know how we should deal with this. Obviously we don't want this to be dependent on ROS, but installing OpenCV from source causes issues if you've already got it. 

Perhaps we should add the OpenCV installation as optional? Like the user can decide whether or not to install it from us from a prompt, or through a flag?

navganti, 2018-01-10:
OR @leokoppel we can check if opencv is already installed on the system. If so, don't install...

leokoppel, 2018-01-13:
We should install to /opt/wavelab for any automated install. This will prevent our install from conflicting with other packages, e.g. OpenCV packaged with ROS.

The error `'CAP_IMAGES’ is not a member of ‘cv’` is a special case when opencv2 is installed in `/usr`, not by ROS. For this special case I don't know any workaround.

As I wrote elsewhere:

> This issue is observed when opencv2 is installed in /usr/include/opencv2. One way to resolve it is to uninstall opencv2.
> 
> It seems OpenCV does things like #include "opencv2/core/mat.hpp" and I don't know a way to preempt /usr/include/opencv2/core/mat.hpp being found first, over the one in /opt/ros. Surely there must be a way to have opencv2 and 3 installed at the same time?