wavelab/libwave #289: Throttle insufficient angular slice warning

mpitropov, 2018-07-13:
#### Platform
Ubuntu 16.04, Master

#### Steps to Reproduce
1. Run occupancy grid on the car or on bag
2. See a lot of warnings

https://github.com/wavelab/libwave/blob/efca86c2dbc006dd64dbb0545390073fe5d30b6e/wave_matching/include/wave/matching/impl/ground_segmentation.hpp#L355

#### Expected result
Not as many warnings or no warnings.
#### Observed result
Lots of warnings.
#### Possible cause
Model is not sufficient for some reason.
#### Related issues
None that I know of but it sounds pretty bad.

rowandempster, 2021-05-02:
Hey @mpitropov, any resolution to this? We (WATonomous at UW) are running into this log and seems to be preventing the Moose occupancy grid module from creating a map (at least we don't see one being published).

mpitropov, 2021-05-02:
This shouldn't be stopping the occupancy grid module from creating the map. The fix is pretty easy for this though but I never got the time to add it, test the fix and open a PR.

If you want to check if this is the issue then the fix I was going to implement was just changing the line to an info message that only prints every 10 seconds. http://wiki.ros.org/roscpp/Overview/Logging
`ROS_INFO_THROTTLE(10, "WARNING:Insufficient Model for angular slice");`

closed 2021-06-30 by mpitropov